climent/atreyu #2: gerbers files for V2 (none-choc)

maxyurk, 2023-05-19:
Hi

First of all I'd like to thank you for this awesome open source keyboard.
I'm interested in building one.
My ask is if it would be possible to include the gerbers for the V2 under the sources/releases. I'm not experienced with KiCad and when I tried to create them I got an issue I couldn't resolve on my own.

Also if you're interested I can help out creating a github page for this project. I noticed a stale branch for it in the repo. https://docusaurus.io/ is a pretty good option I'm familiar with.

Thanks in advance.

climent, 2023-05-21:
Gerber files for v2 have been added to the repo. Enjoy!

closed 2023-05-21 by climent

maxyurk, 2023-08-04:
Thank you. It turned out great. Really happy with it.
[image]

climent, 2023-08-04:
Looks really nice! Hope you enjoy it!