climent/atreyu #3: Rotary Encoder Troubleshooting

TimCummings, 2023-07-08:
Hello, and thanks for the fantastic keyboard design!

This isn't exactly an "issue", I'm just having some trouble getting rotary encoders to work and wanted to see if you had any suggestions. I have an Atreyu v2 choc PCB with a nice!nano and [EC12 encoders](https://tech.alpsalpine.com/e/products/detail/EC12E2440301/). I have the two `CONFIG_EC11` lines uncommented in the top level `atreyu.conf` file. And I can't get the encoders to do anything (I've tried volume +/- and page up/down so far; I have the same key codes on other switches and they work as expected there.)

If it were a switch, I'd check continuity across the pins with the switch pressed, but I'm less sure how to start troubleshooting a rotary encoder.

Also, the underglow status LEDs - are they used at all by default or do I need to add something in the firmware for those?

climent, 2023-07-11:
Let me figure it out and i will get back to you.

climent, 2023-07-14:
First of all, for the encoders to work with a nice!nano board, the 3 extra pins in the middle of the board need to be connected. Did you connect those?

I am right now testing support for the encoder on the right, as I had only configured the code for the encoder on the left. I will let you know once it is working correctly.

TimCummings, 2023-07-15:
> First of all, for the encoders to work with a nice!nano board, the 3 extra pins in the middle of the board need to be connected.

Is that pins 101, 102, & 107 in the screenshot here? https://nicekeyboards.com/docs/nice-nano/

climent, 2023-07-15:
Correct. The routing of the tracks needs those pins. The default configuration in the zmk-config repository here is set for the encoder on the left side.

https://github.com/climent/zmk-config/tree/5e345b4ee15ec3eaf14a35d0d8089bc64e64358b

After that I made some changes that broke the board, but I have not had the time to continue investigating.

TimCummings, 2023-07-15:
That was the issue: I overlooked those three pins. Sorry about that; thanks for your help!

The left is working great; the right works but is a little funky (seems to send both presses with a slight delay in the middle.) I'll fiddle with it some and comment back if I have any success. I was just stuck when I couldn't get either of them to do anything - which makes sense if they weren't even connected to the controller! :facepalm:

closed 2023-07-15 by TimCummings