climent/atreyu #4: Is a build guide still in the works?

Richiban, 2024-01-29:
I've started a build of the Atreyu and it's the first keyboard I'll be building from scratch.

I've got the PCB and have soldered on the hotswap sockets and the socket for the controller; I'm a bit lost, however, with everything else. I've got a nice!nano because I'm planning a wireless build but I'm happy to stick with wired to get it up and running and will upgrade it later. Apologies if these questions are answerable by looking at the KiCad files, but that's rather beyond my ability at the moment.

For the wired version I need to know:
* Where should I put the reset switch? No switch is visible in the photos of the complete keyboard.
* What kind of diodes do I need? The PCB seems to have symbols for SMD diodes on one side of the board and through-hole on the other. Do I need both?
* The socket for the controller seems to have more pins than exist on the nice!nano (i.e. there's one shorter bank on the "bottom" of the controller that the nice!nano doesn't have). Am I giving anything up using nice!nano?

Eventually, to make it wireless, I need to know:
* Where do I put the battery switch?
* Where do I put the battery connector?

Thank you very much for releasing this keyboard design; it's by far my favourite I've ever seen! I currently have a pre-built Sofle that I love, but I'm looking to go unsplit / unibody.

climent, 2024-01-29:
Hi Richard.

First of all, thanks for reaching out. I understand building your first
keyboard may be a daunting task, and i will try to help you as much as i
can.

What version of the PCB did you use? The latest one (2.0.1) should be the
one to use, as it fixed a couple of bugs on the previous ones. Can you
confirm you have that one?

For the reset switches, there are 2 options: a lateral push switch,
available at
https://splitkb.com/collections/keyboard-parts/products/reset-buttons, or a
flat switch, like this one: https://www.digikey.ie/short/ntjm0mnn

The side button goes here:

[image: Screenshot 2024-01-29 at 14.08.24.png]

While the flat button goes here:

[image: Screenshot 2024-01-29 at 14.08.55.png]

There is also a battery switch, available from here:
https://splitkb.com/products/wireless-controller-expansion-bundle

It goes either here:

[image: Screenshot 2024-01-29 at 14.10.02.png]

Or here:

[image: Screenshot 2024-01-29 at 14.10.21.png]

I recommend using the internal ones, as they are accessible from the 2
holes in the bottom plate, and they look nicer in my opinion.

The diodes are standard 1N4148W signal diodes, available here:
https://splitkb.com/collections/keyboard-parts/products/smd-diodes

They can be soldered on the bottom side of the board, with the side of the
line pointing in the direction of the arrow. The Soufflé build guide
https://github.com/climent/SouffleKeyboard/blob/main/docs/build_guide.md
has some photos on how to set them up.

The 3 extra pins in the middle of the nice nano are for rotary encoders,
but they are currently broken in the ZMK software, so you don't miss much,
right now.

If you want to have the encoders, then you have to move the battery to the
big hole in the board. It has space for a 500 mAh battery.

It goes soldered to the BAT/GND pins here:

[image: Screenshot 2024-01-29 at 14.16.31.png]

Let me know if you have any more questions.

Richiban, 2024-01-29:
Thank you so much for your reply @climent, unfortunately the images have not come through in your comment.

There's no rush, but if you have time to come back and edit your comment with the uploaded images I'd be very grateful.

Thanks!

climent, 2024-01-29:
You can provide an email address to "climent () gmail ! com" and i
can forward you the email with the images.